eclispe/rydeplayer #12: Add signal parameters OSD widgets

eclispe, 2020-07-08:
Add signal parameters OSD widgets, including MER, modulation type, and constellation.
Requies #10

davecrump, 2020-07-14:
Desired default behaviour is that OSD display of MER, modulation type, Service and Service provider should be displayed for about 5 seconds after video start, and then disappear.  It should then be possible to reselect the OSD by pressing a remote control key (Info?) or (for use in repeaters) setting a GPIO pin low.

eclispe, 2020-07-14:
My current plan should be compatible with this if I add an option to autoshow the OSD after lock and add a GPIO pin.
The plan was while the menu is closed to use the existing select button to toggle the whole OSD and the up and down keys to cycle through various OSD modes that changes which and how many of the parameters are displayed, possibly allow these modes to be configured in the config file in the future. Some of these features will be broken out into their own issues once this has progressed further.

davecrump, 2020-07-14:
Excellent!  Provisionally allocated GPIO physical pin 37 on the GPIO listing. [https://wiki.batc.org.uk/Ryde_GPIO_Connections](url)

commit 1c889b7e (2020-11-29): Merge branch 'osd'

closes #12
closed 2020-11-02 by eclispe