eclispe/rydeplayer #14: Raspberry Pi 4 dual screen

eclispe, 2020-07-08:
Add support for running on multiple screens on the raspberry pi 4 and displaying different things on both.

Suggestions on what do do with the seconds screen welcome.

G0MJW, 2020-07-11:
The obvious thing to show is the second TS but assuming that's beyond the PI capability how about showing the WB transponder spectrum and chat?

eclispe, 2020-07-11:
> The obvious thing to show is the second TS

Outputting the second TS would probably work if I longmynd supported it and as long as both streams aren't very high symbol rate.

I also thought about outputting the raw video on the second one, no menu, no OSD.

If there are a few popular options then a setting could be added to select one.

eclispe, 2021-12-02:
85554a55e6729a88094e32655719e2cef4dda7f6 adds the PyDispmanx support for display selection