eclispe/rydeplayer #20: Mute Button

eclispe, 2020-07-08:
Add expose access to VLC's mute function. Include menu item and button bindings.
Must include an OSD widget so depends on #10.

G0MJW, 2020-07-20:
Any chance of adding this to the remote control as well? Most have mute and also volume control.

davecrump, 2020-07-20:
Mike - please provide the codes for those buttons on the SF8008, and we will add them to the config file.  We already have them defined (in readiness for Tim including the functionality) for many of the other remotes.  See https://github.com/davecrump/RydeHandsets/blob/definitions/virgin.yaml

G0MJW, 2020-07-20:
Hi Dave

 

I al logged in now so after the annoyance of trying to copy from SSH using CTRL+C (It’s almost deliberate I am sure) here is the output – buttons in order

 

Testing events. Please, press CTRL-C to abort.

320.200038: lirc protocol(nec): scancode = 0x514

320.200069: event type EV_MSC(0x04): scancode = 0x514

320.200069: event type EV_SYN(0x00).

320.250035: lirc protocol(nec): scancode = 0x514 repeat

320.250067: event type EV_MSC(0x04): scancode = 0x514

320.250067: event type EV_SYN(0x00).

326.710041: lirc protocol(nec): scancode = 0x503

326.710070: event type EV_MSC(0x04): scancode = 0x503

326.710070: event type EV_SYN(0x00).

326.760039: lirc protocol(nec): scancode = 0x503 repeat

326.760067: event type EV_MSC(0x04): scancode = 0x503

326.760067: event type EV_SYN(0x00).

328.060061: lirc protocol(nec): scancode = 0x502

328.060088: event type EV_MSC(0x04): scancode = 0x502

328.060088: event type EV_SYN(0x00).

328.110038: lirc protocol(nec): scancode = 0x502 repeat

328.110066: event type EV_MSC(0x04): scancode = 0x502

328.110066: event type EV_SYN(0x00).

332.980069: lirc protocol(nec): scancode = 0x50e

332.980094: event type EV_MSC(0x04): scancode = 0x50e

332.980094: event type EV_SYN(0x00).

333.030048: lirc protocol(nec): scancode = 0x50e repeat

333.030073: event type EV_MSC(0x04): scancode = 0x50e

333.030073: event type EV_SYN(0x00).

333.750044: lirc protocol(nec): scancode = 0x51a

333.750069: event type EV_MSC(0x04): scancode = 0x51a

333.750069: event type EV_SYN(0x00).

333.800046: lirc protocol(nec): scancode = 0x51a repeat

333.800071: event type EV_MSC(0x04): scancode = 0x51a

333.800071: event type EV_SYN(0x00).

[... 232 lines cut ...]
    ONE:    0x509

	
    TWO:    0x51d

	
    THREE:  0x51f

	
    FOUR:   0x50d

	
    FIVE:   0x519

	
    SIX:    0x51b

	
    SEVEN:  0x511

	
    EIGHT:  0x515

	
    NINE:   0x517

G0MJW, 2020-07-20:
Doing the update from the dev version – it’s taking ages doing this sort of thing, is that normal?

 

Removing 'diversion of /boot/overlays/mpu6050.dtbo to /usr/share/rpikernelhack/overlays/mpu6050.dtbo by rpikernelhack'

G0MJW, 2020-07-20:
SF8008 - NEC protocol

  | POWER: 0x514
  | UP:       0x503
  | DOWN:   0x502
  | LEFT:   0x50e
  | RIGHT:  0x51a
  | SELECT: 0x507
  | BACK:   0x513 (There is no   back – this is called EXIT)
  | MENU:   0x500
  | MUTE:   0x51c
  | VOL+:   0x541 
  | VOL-:   0x542
  | CHAN+:  0x5ef
  | CHAN-:  0x5ee
  | ZERO:   0x512
  | ONE:      0x509
  | TWO:      0x51d
  | THREE:  0x51f
  | FOUR:   0x50d
  | FIVE:   0x519
  | SIX:      0x51b
  | SEVEN:  0x511
  | EIGHT:  0x515
  | NINE:   0x517

eclispe, 2020-07-20:
There are already some hints and placeholders in the code for mute (including handset button), which is the first widget once OSD (#10) is ready.

commit 9507e37c (2020-10-14): Adds video mute

closes #20
closed 2020-11-02 by eclispe