eclispe/rydeplayer #36: Add functionality to scan for correct SR

davecrump, 2020-07-14:
It would be useful to be able to select "scan for SR" where the receiver would listen on a single frequency, but continually try (sequentially) a number of predefined (in a config file) SRs until it achieved lock.  This would mean that a single receiver used in a repeater could accept input signals transmitted using any of a predefined list of SRs.  A typical list of SRs would be 2000, 1000, 333 and 125.

eclispe, 2020-07-14:
Longmynd currently requires restarting to change the symbol rate which can take a few seconds (depending on hardware). It may be better to add support for this or a more interactive interface to longmynd first.

davecrump, 2020-07-16:
This might still be practical without changing LongMynd.  A re-scan period of 10 seconds would be acceptable, with preferably 4 scanned SRs.  Even 2 scanned SRs would be useful.

eclispe, 2020-07-16:
With a rescan period of 10s and 4 symbol rates to try it could take over 30s to lock on to a signal which is possibly a bit long or at least would require displaying which SR its currently scanning on the no lock screen. If my understanding of how the nim works is correct you could theoretically scan for 2 symbol rates simultaneously, but again that is going to require some longmynd modifications.

davecrump, 2020-08-11:
Longmynd will now scan for multiple SRs (and frequencies) if given a list of up to 4 SRs (and/or frequencies) separated by commas (no spaces).  The Ryde config.yaml file needs to be mdified so that it can handle multiple SRs and frequencies, and then these multiple values need to be passed to Longmynd in the correct format.  Perhaps implement in 2 stages: Stage 1, multiple values set manually (or by console menu) in config file, and handled unchanged by Ryde Player.  Stage 2, allow these multiple values to be changed/set from the Ryde menu system.

eclispe, 2020-08-12:
I'll use this for stage one tracking and we can file a new issue for full GUI support later.
Will still need some GUI modifications to add placeholders to prevent the menu from crashing.