eclispe/rydeplayer #37: Add Compatibility with Serit FTS-4335 NIMs

davecrump, 2020-07-15:
The LongMynd software running stand-alone will function with either the current Serit FTS-4334 NIM, and the older FTS-4335 NIM.  The Ryde software currently displays the following error when started with an FTS-4335 NIM:

State.Stopped
Flow: main
      Status: Main Frequency=741500 KHz
              Main Symbol Rate=1500 KSymbols/s
              Using First Minitiouner detected on USB
              Main TS output to FIFO=/home/pi/lmmedia
              Main Status output to FIFO=/home/pi/lmstatus
              Main refers to TOP F-Type
Flow: Fifo Init
      Status: opened fifo ok
Flow: FTDI init
Flow: FTDI USB init
Flow: FTDI set mpsse mode
Flow: FTDI USB init
Flow: FTDI set mpsse mode
Flow: FTDI setup io
      Status: MPSSE Synched AA FA
Flow: FTDI nim reset
Flow: FTDI GPIO Write: pin 0 -> value 0
Flow: FTDI GPIO Write: pin 0 -> value 1
Flow: Fifo Init
      Status: opened fifo ok
Flow: No-lock timeout, re-init config.
Flow: NIM init
Flow: STV0910 init
Flow: stv0910 init regs
      Status: STV0910 MID = 0x51, DID = 0x20
Flow: STV0910 set MCLK
Flow: Setup equlaizers 1
Flow: Setup carrier loop 1
Flow: Setup timing loop 1
Flow: Tuner init
Flow: Tuner cal lowpass
Flow: Tuner set freq
      Status: tuner:0, f_vco=0x2d41f0, icp=0x2, f=0x2eeee, n=0xc5,
              rdiv=0x1, p=0x1, freq=741500, cfhf=2331
Flow: LNA init 1
ERROR: i2c read reg8 0xce, 0x00
ERROR: lna read 0xce, 0x00
      Status: found an older NIM with no LNA
Flow: LNA init 2
ERROR: i2c read reg8 0xc8, 0x00
ERROR: lna read 0xc8, 0x00
      Status: found an older NIM with no LNA
Flow: FTDI GPIO Write: pin 4 -> value 0
Flow: STV0910 start scan
State.Stopped

About 100 FTS-4335 NIMs were sold to BATC Members by the BATC shop (before it was decided to stock the current FTS-4334).  The difference is, as reported in the error message, that the FTS-4335 NIM does not have an input LNA on either input; beyond that stage it is almost identical.

As these NIMs were purchased by many of our keener members (the early adopters) it would be very useful if they could be made to work with the Ryde software.

davecrump, 2020-07-15:
Although this commit handles the different status message, it does not handle the "spurious" error messages that the longmynd throws up when using an FTS-4335 NIM.  So, the resolution is either to change longmynd to not output these specific error messages (the neater solution), or change Ryde to handle these specific error messages.  The pertinent extract from stdout when using an FTS-4335 NIM is:

Flow: LNA init 1
ERROR: i2c read reg8 0xce, 0x00
ERROR: lna read 0xce, 0x00
      Status: found an older NIM with no LNA
Flow: LNA init 2
ERROR: i2c read reg8 0xc8, 0x00
ERROR: lna read 0xc8, 0x00
      Status: found an older NIM with no LNA
Flow: FTDI GPIO Write: pin 4 -> value 0
Flow: STV0910 start scan
State.Stopped

These error messages cause ryde to stop.

One of the error messages is generated here: [https://github.com/eclispe/longmynd/blob/master/nim.c#L110](url) and the other here [https://github.com/eclispe/longmynd/blob/master/ftdi.c#L399](url)

eclispe, 2020-07-15:
Ignoring or suppressing the "lna read" errors should be fairly easy, have to be careful ignoring the lower level i2c error as other legitimate errors can generate similar messages.

davecrump, 2020-07-17:
Not quite sorted yet.  It does work, but seems to need other remote control inputs to enable it.

Ian reports: 

I can report some success with the old serit tuner.
When it first boots up it does show the not loaded screen
But if you change LNB input from top to bottom to top it bursts into life.
If you change frequency, again it sits on not loaded until you toggle the input
Once it is going it seems to be fine on a fixed channel - ie if a new signal comes along it decodes - as long as you don't change channel or SR - it then stalls.
Restart LongMynd doesn't seem to do anything
On the new serit it all works as expected
To view my file, click this link:
[https://1drv.ms/v/s!AktofuOcubWsrVK_F_NgdPIwK-uQ
](url)
The above video shows the issue. Unfortunately at the time there were a couple of very loud stations but you can see it would have worked

and Colin says: 

Not seeing the same issue.  Boots straight up to the beacon but seems necessary to set the LNB Polarity to horizontal to RX other stations. 

Might need to try to lend you an old tuner!

eclispe, 2020-07-17:
That is very weird, I suspect my error ignoring code is has bugs or something. It probably dumps out info when it crashes that might give some insight. I wonder if there is a sensible/secure way for me to login to a pi remotely, but hands on a tuner would also work.

commit 1585551e (2020-07-15): Add support for FTS-4335 NIM

Fixes #37
commit f61060b1 (2020-07-16): Add support for FTS-4335 NIM

Should fix #37
commit 549a767c (2020-07-20): Speed up STDOUT reading to fix race condition

If you don't read from longmynd fast enough STDOUT locks up and you can't detect when its done loading.
Fixes #37
closed 2020-07-20 by eclispe