eclispe/rydeplayer #38: 4k Monitor Compatibility

G8YTZ, 2020-07-16:
When booting with a 4k monitor attached to either monitor port, SSH access fails and the receiver software fails to start. Carrying out a port scan port 22 is not listening. Whilst booting the resolution mode changes. RPI 4 2GB.

davecrump, 2020-07-16:
Justin - Please could you see what behaviour you get with a bare SD Card withour Ryde loaded (just Raspios Lite and the ssh file)?  Can you then ssh in with the 4K monitor attached?  If there is a problem with creating a user session (as you report) then Ryde will not start up.  Not a Ryde problem, but a basic Raspberry Pi problem possibly.  Difficult for us to test without a 4K monitor.

G8YTZ, 2020-07-16:
Hi Dave, so I did that, no issue with the OS. Actually this time a slightly different result, but I have upgraded the Ryde Player s/w since the first go last night. Basically SSH now works in both monitor scenarios, with 4k monitor and with 1k monitor, however with the 4k monitor the Ryde software does not start, just ends at the system prompt. On the 1k monitor no problem the software starts and received just fine. On the 4k monitor the Rydeplayer software will not start, even if stopped/started via the menu.

Monitors are both 2019 models: 
4k LG 27UK850 3840 x 2160
1k LG 34WK650 2560 x 1080

Justin.

eclispe, 2020-07-16:
I was wondering if it would work on a 4k display, and who would be the first to try it, slightly disappointed it didn't work. I don't have access to a 4k display to test it on unfortunately so I may need you to try a few things to get this fixed.
Can run it manually and reply with the errors? It should just be a case of doing thins from the system prompt (Dave: correct me if there is a better way to get the errors):
```
cd /home/pi/ryde
python3 -m rydeplayer /home/pi/ryde/config.yaml
```

G8YTZ, 2020-07-16:
4k monitor Connected:

justincockett@Justins-MacBook-2 ~ % ssh pi@192.168.0.231
pi@192.168.0.231's password: 
Linux raspberrypi 4.19.118-v7l+ #1311 SMP Mon Apr 27 14:26:42 BST 2020 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Fri Jul 17 00:32:09 2020

SSH is enabled and the default password for the 'pi' user has not been changed.
This is a security risk - please login as the 'pi' user and type 'passwd' to set a new password.

Wi-Fi is currently blocked by rfkill.
Use raspi-config to set the country before use.

pi@raspberrypi:~ $ cd /home/pi/ryde
pi@raspberrypi:~/ryde $ python3 -m rydeplayer /home/pi/ryde/config.yaml
pygame 1.9.4.post1
Hello from the pygame community. https://www.pygame.org/contribute.html
Namespace(**{'Config File': '/home/pi/ryde/config.yaml'})
/home/pi/ryde
Handset has no button definitions
Handset has no button definitions
  Frequency: 1746000
Symbol Rate: 2000
   Polarity: PolarityEnum.NONE
       Port: inPortEnum.TOP

1k monitor connected

See attached movie on BATC Facebook.

eclispe, 2020-07-17:
What the 1k display does is what its supposed to do, spew a load of debugging everywhere :)
On the 4k screen can you try this:
```
cd /home/pi/pydispmanx
python3 demo.py
```
and post the output and if anything flashes on the display (it might be very quick)?

G8YTZ, 2020-07-17:
4k Monitor:

justincockett@Justins-MacBook-2 ~ % ssh pi@192.168.0.231
pi@192.168.0.231's password: 
Linux raspberrypi 4.19.118-v7l+ #1311 SMP Mon Apr 27 14:26:42 BST 2020 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Fri Jul 17 10:43:23 2020

SSH is enabled and the default password for the 'pi' user has not been changed.
This is a security risk - please login as the 'pi' user and type 'passwd' to set a new password.

Wi-Fi is currently blocked by rfkill.
Use raspi-config to set the country before use.

pi@raspberrypi:~ $ cd /home/pi/pydispmanx
pi@raspberrypi:~/pydispmanx $ python3 demo.py
pygame 1.9.4.post1
Hello from the pygame community. https://www.pygame.org/contribute.html
(3840, 2160)

G8YTZ, 2020-07-17:
1k Monitor: (note: a large blue dot appears in the centre of the screen during this process).

justincockett@Justins-MacBook-2 ~ % ssh pi@192.168.0.231
pi@192.168.0.231's password: 
Linux raspberrypi 4.19.118-v7l+ #1311 SMP Mon Apr 27 14:26:42 BST 2020 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Fri Jul 17 10:49:11 2020

SSH is enabled and the default password for the 'pi' user has not been changed.
This is a security risk - please login as the 'pi' user and type 'passwd' to set a new password.

Wi-Fi is currently blocked by rfkill.
Use raspi-config to set the country before use.

pi@raspberrypi:~ $ cd /home/pi/pydispmanx
pi@raspberrypi:~/pydispmanx $ python3 demo.py
pygame 1.9.4.post1
Hello from the pygame community. https://www.pygame.org/contribute.html
(2560, 1080)
27.90058585653247
no surface
no layer
pi@raspberrypi:~/pydispmanx $

eclispe, 2020-07-17:
Thanks for that, defiantly a display interface problem. Can you try increasing the video memory split using `sudo raspi-config`?
Its under `Advance Options`, `Memory Split`. The maximum value is 512 so start there and see if demo.py works.

G8YTZ, 2020-07-17:
OK Tim, that's a fix. It works ok at 128k on a fairly static 720p picture from GB3JV. It has run now for some hours.

eclispe, 2020-07-17:
excellent thanks, the default is 76 so it didn't need that much more.

Dave: probably no harm in doing this for everyone in the config.txt `gpu_mem=128`

davecrump, 2020-07-20:
ryde-build release 202007200 (Commit c326612 on the BATC GitHub) cures this by adding "gpu_mem=128" to /boot/config.txt.  This issue can now be closed.

closed 2020-07-20 by eclispe

G8YTZ, 2020-07-21:
Thanks Dave & Tim, I've updated the system, I'll test later today. BTW I was over at Dave G8OUX's place yesterday afternoon and we also checked [ok] on a LG 4k TV set. I'll also checked k on my Sony 4k set today.

G8YTZ, 2020-07-21:
I've updated the Pi and all seems to be ok with 4k now. Just one thing I noted, I had modified the config.yaml file for the frequency and symbol rate of the GB3JV repeater in the default section, the update overwrote my custom file, not an future an issue as such as it takes 30s to fix, but it would be nice if the config file was not overwritten during the update process in future releases.

davecrump, 2020-07-21:
Pleased that works.  As explained in the release note [https://forum.batc.org.uk/viewtopic.php?f=130&t=6773](https://forum.batc.org.uk/viewtopic.php?f=130&t=6773) the config file has undergone major changes so I was unable to preserve it on this occasion.

G8YTZ, 2020-07-21:
Sure, Dave its very difficult giving feedback on work that people have done as it almost seems you're criticising the authors all the time whenever a bug is found or an enhancement is asked for. Of course it's supposed only to be constructive, but I guess it's a fine balance. To be absolutely I really am delighted with this software and really appreciate the work you've put into this and the other platforms you've conceived, I just can't wait for the next drop!