eclispe/rydeplayer #40: Enhancement: Support I2C OLED Display on Minitiouner

G8YTZ, 2020-07-17:
It would be nice to have the I2C OLED display supported as per the Minitiouner Windows Software, or perhaps a larger I2C display.

davecrump, 2020-07-17:
The I2C OLED display is driven by the Windows MiniTioune software, and is not appropriate for this simple set-top-box.  An OSD should be available in later Ryde versions with similar functionality.

eclispe, 2020-07-17:
I'm sure there is much value in this once the OSD is working, see #10, #11 and #12

G8YTZ, 2020-07-17:
Dave, my thought is for a confirmation display for RSSI and MER for the front panel for the repeater receiver, but it would also be useful for a home receiver. 

The other alternative would be a split screen option with the OSD on one screen, so a small monitor could be incorporated into the front panel.

I would have thought an I2C would be fairly easy in software, be great if it could display for each tuner output too. 

Justin. 

www.g8ytz.com

eclispe, 2020-07-17:
The Pi does have a I2C in the GPIO however we have other plans for that at the moment. I'm not fundamentally opposed to the idea so I'm not closing this for now but there are some other things that need doing before we can really assess the value/practicalities.

davecrump, 2020-07-17:
Given that most repeater receivers spend their time unattended, I think that the solution proposed at [](https://github.com/eclispe/rydeplayer/issues/12)  should be good enough.  Note that the Windows MiniTiouner solution uses the USB channel to send the I2C data.  We can leave this issue open, but there are many higher priority features to add first.