eclispe/rydeplayer #42: The IR Protocol nec32 is not correctly handled in ir.py

davecrump, 2020-07-17:
Apple TV remotes use nec32 protocol, which is handled with the 'nec' protocol setting of ir-keytable.

The IR code for Menu on my example remote is 0x87eede02.  This seems to be correctly read from the handset.yaml file and is entered in the keytable (line 45 of ir.py) as 2280578562 (which is 0x87eede02 in decimal).

However,  when the handset key is pressed, event.value (line 244 of ir.py) has the value -2014388734.  I suspect that it is an overflow of some sort as 2280578562+2014388734=4294967296 which is 0x19999999.

Low priority.

G0MJW, 2020-07-20:
I noticed my Octagon remote is a little unreliable also - perhaps this is an issue. I found that when testing with the utility sometimes pressing a new button initially gave the previous button data. I assumed this was just a bug in the testing utility but maybe it is something more, like an uninitialised variable perhaps.

davecrump, 2020-07-20:
The testing utility is directly displaying the output of the Linux "ir-keytable -t" command, so very unlikely to be buggy.  I would suspect your remote.  The nec32 bug is deterministic, not random.