eclispe/rydeplayer #44: Facility to change the Red Screen "Not Locked" for user-specified image file/slideshow list of files

G8YTZ, 2020-07-22:
Hi Dave, Tim. One question... The red screen that says "Not Locked" Is this an image file? If so where is it and what's its name? Any chance I can change this for a Test Card file I have with the repeater ident? Even better if it could play a slideshow in a later release on an "unlocked" mode. That might be simpler than external HDMI switching to another source.

eclispe, 2020-07-22:
It isn't an image file, the text is rendered live. There is no reason a slideshow couldn't be added in the future but don't get your hopes up about it being soon.

G8YTZ, 2020-07-22:
Even just a JPG etc. file that I could substitute instead (showing the Test Card and Repeater ID) would be fantastic, slide show can come later.

davecrump, 2020-07-22:
Justin - this is way outside the scope of the Ryde requirement, which is that the Ryde (primarily a simple set-top box) should be capable of being used as one of a repeater's receivers, and have the fact that its output is valid indicated by the change of logic level on a GPIO pin (which is already available in this release).  There is a separate project (albeit not as far advanced and only being discussed amongst those actively involved) to use the LongMynd functionality as part of a digital path, digital only, repeater.  Alternatively, you could use the Ryde's logic signal valid indication to switch an HDMI switch to a caption generator/slideshow (from another RPi?).

G8YTZ, 2020-07-22:
I though it would be a fairly easy enhancement.