eclispe/rydeplayer #57: Remotely configuring scan parameters

m0dts, 2021-06-18:
It would be useful to be able to remotely configure the scan parameters using the network interface.
Extra parameters in the band config would be adequate.

I am working around this now with a loop of tune commands but it is much slower than the internal longmynd scan.

eclispe, 2021-06-19:
The ts timeout is the only scanning parameter that isn't currently accessible via the network interface. Some of the changes that are have been made for DVB-T support but not pushed yet should make moving this one into the bands (or even presets) fairly easy.

m0dts, 2021-06-19:
Thanks for the response, it actually made me look a lot harder into the code, i figured it out eventually :-)
I had not spotted that i could send a list of frequencies/sr's in the tune command.
Excellent!

eclispe, 2021-06-19:
The network uses the same parser as the config file so anything that can be set in a preset can be set over the network.