eclispe/rydeplayer #63: IR handset improvement

mcafzap, 2021-12-19:
At present, adding a new handset is bad enough, but having to submit your new one in order for it not to be over-ridden is less than ideal. Can this be avoided by placing a location anywhere in the user directory and setting this path within the 'menu' program?

mcafzap, 2021-12-19:
I see what I've proposed is probably possible by editing the config file...

closed 2021-12-19 by mcafzap

davecrump, 2021-12-19:
As, you say, this is already possible in rydeplayer by setting the the ir: libraryPath: parameter.  I believe your issue is actually with ryde-build [https://github.com/BritishAmateurTelevisionClub/ryde-build](url) where updates over-write the IR handset directory with the latest handset library.

This was a concious design decision to encourage the submission of handset codes to add to the community library.

mcafzap, 2021-12-19:
Yes, but you wouldn't want my particular 'handset' adding, ha, ha.