eclispe/rydeplayer #66: Crash when received Service Info contains high Unicode characters

davecrump, 2023-01-04:
Python crashed out to the command line while receiving a station with the following service info:

state:{'state': 4, 'provider': 'IU4GAV', 'service': '\x15🍺Andrea🍺', 'modcode': 8, 'pids': {256: 36, 257: 15}}

The significant error message was:

UnicodeError: A Unicode character above '\uFFFF' was found; not supported

Sorry I did not catch the whole trace from the crash.

Dave

dl9sec, 2023-01-07:
Hi Dave,

that sounds for me as the same issue I described at the Ryde forum (https://forum.batc.org.uk/viewtopic.php?f=130&t=7986&sid=f02f38d4f7603588129dd40836f4c319&start=10#p32181).

Where did you catch the error message, which log or output?

73, Thorsten

davecrump, 2023-01-07:
Hi Thorsten

I think that there are 2 issues here - this was definitely a problem with with the Service info containing unsupported unicode characters (emojis).
I captured the error message by running the Ryde in "Ryde Test" mode from the Utils menu on the SSH console menu.
Dave

eclispe, 2023-01-09:
This is a limitation of PyGame that is used for all the onscreen graphics, the version we are using only supports UCS-2 chars which are up to FFFF, (see https://www.pygame.org/docs/ref/font.html). PyGame versions newer than 2.0.3 don't have this issue, Debian isn't including a new enough version until the next release but Pi OS may include a newer version, does anybody have a pi running the newest Pi OS to check which version it comes with?

davecrump, 2023-01-09:
I've just checked on a fully-updated RPi Bullseye release and sadly it's still Version: 1.9.4.post1
Is there a way to pull in a newer version?

eclispe, 2023-01-09:
Theoretically we could pin it in from Bookworm but from doing that sort of thing for other packages in the past I can be a bit of a dependency nightmare. It is probably better if I look at some workaround options first.

dl9sec, 2023-01-09:
I am currently running Ryde on Buster (lite), there is no PyGame installed...

davecrump, 2023-01-09:
Hi Thorsten  PyGame is installed and used by Ryde on Buster Lite.  Try running "pip3 show pygame" and you will find it there.  Dave

commit c4190754 (2023-01-09): Change font rendering to use freetype shim

Allows font rendering beyond U+FFFF
This will probably need to be reverting for pygame 2.0.3 onwards
Fixes #66
closed 2023-01-09 by eclispe